UVVM/UVVM #187: bitvis_vip_gpio   C_VVC_CMD_DATA_MAX_LENGTH   limit

jorlovicBG, 2022-10-08:
Hello,

we are working with bitvis_vip_gpio in our verification flow a lot! We use it mostly when verifying our units to configure the unit using "vector" configuration or readout specific status information. 
Often is the configuration a records with a lot of logic vector parameters. In this case, because of the limited C_VVC_CMD_DATA_MAX_LENGTH we are forced to instantiate a lot of GPIO VVC instances in our harness, regardless of the need of a single point configuration.

We are able to branch our own bitvis_vip_gpio with a modified C_VVC_CMD_DATA_MAX_LENGTH, but I would like to ask a general question: Is it possible to increase this value in your master releases to something more significant (e.g. 512 or 1024).

I believe, there is no issue, if this parameter is adapted, and the backward compatibility should work without any problems.

Thank you!

EspenTallaksen, 2022-10-13:
Hi,

I see your point. Let us check if this has other consequences. If not - it should be possible to implement this in a release very soon.

-- Espen

EspenTallaksen, 2022-10-18:
We will check closer. Our worry is that this might affect other VVCs / VVC-fetches, but we haven't tried it out yet. Hopefully we will have time to consider this in our next meeting on Monday.

- Espen

UVVM, 2023-03-10:
Fixed in release v2023.03.10

closed 2023-03-10 by UVVM