UVVM/UVVM #190: Support terminating commands in SPI VVC executor

am9417, 2022-11-28:
There should be possibility to terminate (or timeout) at least the slave-side commands in bitvis_vip_spi.

am9417, 2022-11-28:
#191

UVVM, 2022-12-02:
Great suggestion.
Could you please change the procedure parameter from terminate_loop to terminate_access, as your pull request allows terminating throughout the access and not just at the start.

am9417, 2022-12-04:
> Great suggestion. Could you please change the procedure parameter from terminate_loop to terminate_access, as your pull request allows terminating throughout the access and not just at the start.

Good to hear. Yes, I renamed it to `terminate_access`, that's better.

UVVM, 2023-04-12:
Entered as pull request. Thus closed here.
https://github.com/UVVM/UVVM/pull/191

closed 2023-04-12 by UVVM