UVVM/UVVM #195: Shared variables must be of protected type

sckoarn, 2023-01-09:
Unsure how others use this but my simulator says that shared variables must be protected types,  2008.

    Shared variables must be of protected type:    

    ../src/adaptations_pkg.vhd:71:19     shared_default_log_destination    t_log_destination    

    ../src/adaptations_pkg.vhd:432:19    deprecated_subprogram_list        t_deprecate_list

EspenTallaksen, 2023-01-10:
As "ordinary" shared variables were very useful and very user friendly these have been kept in many testbenches and testbench environments after being removed from the VHDL LRM. This is to my knowledge also the reason why all simulators (at least Modelsim/Questa, Riviera Pro/Active HDL, Vivado and GHDL) are allowing these shared variables - but often with an alert that they are deprecated. You may normally turn this alert off, which is absolutely fine when this is still working perfectly well in the simulator.
In UVVM_3_BETA this has been fixed, but we recommend to keep on using the default UVVM for some time, as UVVM_3_BETA is a beta version - mainly because we are in the progress of adding more important updates there which are not yet properly verified or completed.

sckoarn, 2023-01-12:
I understand the state of things.  I am verifying a new simulator and we encounter these devotions from the 2008  LRM constantly.  If in fact the package is to be2008  compatible, it should meet the LRM requirements.  There are other packages available, I will not name them as you already know them,  they are compliant.

We intend to keep our simulator as compliant with the LRM as possible, and provide switches for users to make the simulator act like the others.

I am in fact not using your package to do verification of RTL, but of the implementation of VHDL in our simulator.
for reference,  this is the simulator.  https://metrics.ca/

The package is providing interesting cases for us,  Thanks for producing it :)

EspenTallaksen, 2023-01-30:
Sounds interesting.
Let us know if you need any info in the future.
To your information - UVVM is great for constrained random testing (which uses a lot of CPU power), but it is also great for directed - far more CPU efficient testing as well.
Best of luck with your tool.

closed 2023-04-12 by UVVM