UVVM/UVVM #196: [QUESTION/POSSIBLE USER ERROR]: AXI-Stream VVC byte_endianess not working?

sibeov, 2023-01-20:
# Explanation and question
How does the `byte_endianness` work with regards to the `t_axistream_bfm_config` and its associated VVCs?
It seems to have no effect on the output and input of the VVCs whenever I change it?
I have tried both `FIRST_BYTE_LEFT` and `FIRST_BYTE_RIGHT` (the others as well actually), but there is no change in byte order in Modelsims wave window?
It's probably a user-error on my part or something I have misunderstood.

# How to reproduce
Instantiate a "tx" and "rx" VVC, connect to a DUT (can be just a FIFO or a dummy black box transmitting the input to the output) and try all `byte_endianness` possibilities.

# Solution for now
Implemented a "swap" function which swaps the bytes to desired byte order.

mariuselv, 2023-02-25:
Hi,

Thank you for your message. We appreciate your report and our team will investigate the issue promptly.

Best regards,
Marius

erick166, 2023-04-17:
Hi,

The config **byte_endianness** in AXI-Stream has 2 possible values: **LOWER_BYTE_LEFT** and **LOWER_BYTE_RIGHT** (_FIRST_BYTE_LEFT_ and _FIRST_BYTE_RIGHT_ are the same and will be removed, while _LOWER_WORD_LEFT_ and _LOWER_WORD_RIGHT_ are not used here).

The purpose of this config is to change the order of the bytes inside the array elements of a t_slv_array, e.g.

```
variable v_data_array_4_byte  : t_slv_array(0 to 1)(31 downto 0);

v_data_array_4_byte := (x"11223344", x"AABBCCDD");
axistream_transmit(AXISTREAM_VVCT, C_MASTER_IDX, v_data_array_4_byte, "4 bytes transmit");
```

With LOWER_BYTE_LEFT config, the order will be:

```
# UVVM: ID_PACKET_DATA                   125.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"11", byte# 0. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   125.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"22", byte# 1. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   125.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"33", byte# 2. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   125.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"44", byte# 3. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   142.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"AA", byte# 4. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   142.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"BB", byte# 5. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   142.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"CC", byte# 6. '4 bytes transmit'  [10]
# UVVM: ID_PACKET_DATA                   142.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"DD", byte# 7. '4 bytes transmit'  [10]
```

Whereas with LOWER_BYTE_RIGHT, the order will be:

```
# UVVM: ID_PACKET_DATA                   175.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"44", byte# 0. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   175.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"33", byte# 1. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   175.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"22", byte# 2. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   175.0 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"11", byte# 3. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   192.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"DD", byte# 4. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   192.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"CC", byte# 5. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   192.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"BB", byte# 6. '4 bytes transmit'  [16]
# UVVM: ID_PACKET_DATA                   192.5 ns  AXISTREAM_VVC,2                axistream_transmit(8 words[8b])=> Tx x"AA", byte# 7. '4 bytes transmit'  [16]
```

Is this how you intended to use it?

sibeov, 2023-05-31:
Hi.

Thanks for the response and sorry for late response. A lot going on.
Will follow up and check out suggestions.

Kind Regards

erick166, 2023-10-26:
Hi, any updates on this issue?

sibeov, 2023-10-30:
> Hi, any updates on this issue?

Hi. Have not had time. Sorry. Will check it out this week.

sibeov, 2023-11-13:
I 

> # UVVM: ID_PACKET_DATA                   192.5 ns  A

Yes. This was the intention. But I did not seem to change the output as I would have expected. I have not been able to check it out any further since I mentioned the issue. But I was able to test it in the latest version and it worked as expected. I do not mind closing this issue. If anything I can reopen it at a later stage.

Sorry for the very late reply.

Kind Regards

erick166, 2023-11-14:
Alright, then we can close this issue. Feel free to reopen it if needed.

closed 2024-01-15 by UVVM