UVVM/UVVM #197: Awaiting specified command index with AXI VVC

am9417, 2023-02-01:
Is it possible to await a specified command index with AXI VVC? I'm trying to do:
```vhdl
axi_read(....);
cmd_idx := get_last_received_cmd_idx(AXI_VVCT, AXI_VVC_IDX);
await_completion(AXI_VVCT, AXI_VVC_IDX, cmd_idx, timeout, "awaiting.");
```
but I get TB_ERROR with
> await_completion with a specified wanted_idx is not supported by AXI_VVC.

Reason for the error seems to be in [UVVM/bitvis_vip_axilite/src/axilite_vvc.vhd, line 206](https://github.com/UVVM/UVVM/blob/master/bitvis_vip_axi/src/axi_vvc.vhd#L206). Why is this so? Is there an alternative way to await the specified command index with AXI VVC?

For example, I could have following sequence
```vhdl
axi_read(...);
cmd_idx := get_last_received_cmd_idx(AXI_VVCT, AXI_VVC_IDX);

for i in x to very_big_number loop
    axi_write(...);
end loop;

await_completion(..., cmd_idx, "Wait only until the read is complete");
do_something_after_read_finished;
```
But obviously this is not supported, and I have to await for all commands (irrelevant commands aswell) in the command queue to get completed, right? Then I cannot issue `do_something_after_read_finished` when I would've wanted to, i.e. right after the read has finished. Instead, I have to wait for all the AXI VVC commands, including the multiple axi_writes, have finished, too, which may be too late.

mariuselv, 2023-02-25:
Hi,

Thank you for your message. We appreciate your report and our team will investigate the issue promptly.

Best regards,
Marius

UVVM, 2024-09-20:
The AXI VVC has been updated to handle awaiting for a specified command index in the latest release v2 2024.09.19

closed 2024-09-20 by UVVM