UVVM/UVVM #200: Register Wizard

m-kru, 2023-02-24:
What has happened with the Register Wizard? Has it been abandoned? It looks like there is no longer a way to download it.

EspenTallaksen, 2023-02-24:
Do you have the old version?  If so, it was never modified, and no update could be expected.
If you don't have it, I can send the download to you. If so, please send your email to et@emlogic.no

- Espen

m-kru, 2023-02-24:
@EspenTallaksen Do you have any new tool?

closed 2023-02-24 by m-kru

EspenTallaksen, 2023-02-24:
No, I'm sorry. When we made this in Bitvis, it seemed the request for such a tool was dropping - due probably to other solutions popping up from technology vendors and others. That was also why we did not invest in further updates.
-- Espen