UVVM/UVVM #201: Bug: Queue Status Depends on Undefined Process Execution Order

bpadalino, 2023-03-03:
The `nvc` simulator seems to have exposed a bug in the UVVM framework.  Please see [`nvc` issue #614](https://github.com/nickg/nvc/issues/614).

The error happens in `fetch_command_and_prepare_executor` in the `td_vvc_entity_support_pkg` package.  Specifically, if an extra `wait for 0 ns` is added before the check for the empty queue, the behavior changes.

This was also verified to be an issue in the order that processes are run as mentioned [here](https://github.com/nickg/nvc/issues/614#issuecomment-1454008648).  Running them in reverse order fixes the issue without requiring the extra `wait` statement.

Since the VHDL LRM doesn't specify an order for processes to be run, this appears to be a bug in UVVM itself.

mariuselv, 2023-03-13:
Hi Brian,

Thank you, we will have a look at this issue.

Best regards,
Marius

closed 2023-03-21 by UVVM

bpadalino, 2023-03-21:
Not to be a pest, but can you link to the commit which fixes this bug?

mariuselv, 2023-03-22:
Hi,
The issue is fixed in commit 87b9f2b68268de39d537b5828946155088d8b72e

Br,
Marius