UVVM/UVVM #202: vip_axistream fails with error message (SIGSEGV) Bad handle or reference.

chriseberlein88, 2023-03-07:
When using a axistream vvc with an array of input stream modelsim fails to simulate with the error: (SIGSEGV) Bad handle or reference.

The input streams are used as follow:

`type ta_axistream_if is array (natural range <>) of t_axistream_if;
`       
 ```
signal axistream_outputs    : ta_axistream_if(0 to 5)( 
                               tdata(C_AXIS_DATA_WIDTH - 1 downto 0),
                               tkeep(C_AXIS_DATA_WIDTH/8-1 downto 0),
                               tuser(C_AXIS_USER_WIDTH-1 downto 0),
                               tstrb(C_AXIS_DATA_WIDTH/8-1 downto 0),
                               tid(C_AXIS_ID_WIDTH-1 downto 0),
                               tdest(C_AXIS_DEST_WIDTH-1 downto 0));
```

        i_axistream_vvc_output: entity bitvis_vip_axistream.axistream_vvc
        generic map(
        )
        port map(
            clk              => clk,
            axistream_vvc_if => axistream_outputs(0)
        );

I figured out the fix for that behaviour is similar to issue #118 
Fixing it can be done by changing the calls in axistream_vvc transmit/receive/expect calls from

	axistream_transmit_bytes(
		data_array           => v_cmd.data_array(0 to v_cmd.data_array_length-1),
		user_array           => v_cmd.user_array(0 to v_cmd.user_array_length-1),
		strb_array           => v_cmd.strb_array(0 to v_cmd.strb_array_length-1),
		id_array             => v_cmd.id_array(0 to v_cmd.id_array_length-1),
		dest_array           => v_cmd.dest_array(0 to v_cmd.dest_array_length-1),
		msg                  => format_msg(v_cmd),
		clk                  => clk,
		axistream_if         => axistream_vvc_if,
		scope                => C_SCOPE,
		msg_id_panel         => v_msg_id_panel,
		config               => vvc_config.bfm_config);
		

to

	axistream_transmit_bytes(
		data_array           => v_cmd.data_array(0 to v_cmd.data_array_length-1),
		user_array           => v_cmd.user_array(0 to v_cmd.user_array_length-1),
		strb_array           => v_cmd.strb_array(0 to v_cmd.strb_array_length-1),
		id_array             => v_cmd.id_array(0 to v_cmd.id_array_length-1),
		dest_array           => v_cmd.dest_array(0 to v_cmd.dest_array_length-1),
		msg                  => format_msg(v_cmd),
		clk                  => clk,
		axistream_if.tdata         => axistream_vvc_if.tdata,
		axistream_if.tkeep         => axistream_vvc_if.tkeep,
		axistream_if.tuser         => axistream_vvc_if.tuser,
		axistream_if.tvalid         => axistream_vvc_if.tvalid,
		axistream_if.tlast         => axistream_vvc_if.tlast,
		axistream_if.tready         => axistream_vvc_if.tready,
		axistream_if.tstrb         => axistream_vvc_if.tstrb,
		axistream_if.tid         => axistream_vvc_if.tid,
		axistream_if.tdest         => axistream_vvc_if.tdest,
		scope                => C_SCOPE,
		msg_id_panel         => v_msg_id_panel,
		config               => vvc_config.bfm_config);

mariuselv, 2023-03-13:
Thank you,
we will have a fiks soon.

Br
Marius

closed 2023-03-21 by UVVM

chriseberlein88, 2023-03-23:
The Issue seems to be only relevant to specific Modelsim versions.
Updated to ModelSim Microsemi Pro 2022.2 and the failure is gone without any fix.

FY: Version that failed was ModelSim Intel FPGA Starter Edition 2020.1