UVVM/UVVM #203: Avalon ST empty symbols

whiteway11890, 2023-03-07:
I've scoured the source code and the internet with no resolution, so I'm asking/submitting an issue here.

I'm using UVVM to test a system that has both an Avalon ST master and slave implementation. The implementation utilizes the "empty" field, which allows the system to express that some symbols in the word are invalid. I am using a symbol size of 8 bits and a word size of 4 symbols (32 bits). The valid value for "empty" is 0-3.

In the avalon_st_bfm_pkg.vhd file, empty symbols *appear* to be taken care of in the `avalon_st_receive` function (I am calling it via `avalone_st_expect'):
```
-- Check the number of empty symbols on the last data transfer
if c_symbols_per_beat > 1 then
    v_empty_symbols := to_integer(unsigned(avalon_st_if.empty));
end if;
```

However, the function can only end at a multiple of the symbols per beat, `c_symbols_per_beat`, since `v_symbol_array` has a length that is `v_normalized_data'length*c_symbols_per_beat`. By this time, `v_sym_in_beat` must be `c_symbols_per_beat-1` and `v_sym_count` must be a multiple of `c_symbols_per_beat` (-1 for 0 indexing) due to how `v_symbol_array` is allocated. If `v_empty_symbols` is anything but 0, an error is thrown.

```
-- Finish receiving data when the symbol array ends
if v_sym_cnt = v_symbol_array'length - 1 then
    v_done := true;
    -- Check that empty signal is set on the last data transfer
    if v_sym_in_beat /= c_symbols_per_beat - 1 - v_empty_symbols then
        alert(error, v_proc_call.all & "=> Failed. Empty signal not set correctly for the last transfer. " & add_msg_delimiter(msg), scope);
    end if;
end if;
```

For my own testing, I changed `if v_sym_cnt = v_symbol_array'length - 1 then` to `if v_sym_cnt = v_symbol_array'length - 1 - v_empty_symbols then` and it is working great. 

Am I missing something or is this an oversight in the design? There should also be checks to make sure that "empty" is 0 if it's not the last word in a packet.

-----------------------------------------------------------------------------------------

On the flip (transmit) side, a similar issue arises. The `empty` signal is filled, but the function can only exit on a multiple of `c_symbols_per_beat`, and thus `empty` is set to 0 no matter what. I had to add an argument to tell the function how many symbols were acceptable instead of the default `v_normalized_data'length*c_symbols_per_beat`

Again, am I missing something or was this feature not fully developed?

Thanks!

erick166, 2023-04-18:
Hi,

I assume you are using a data_array with an element size of 32 bits when calling the transmit and receive procedures, is that correct?

It might not be clear from the documentation, but when using packet-based transfers, you should use a data_array with the symbol size elements, i.e. `t_slv_array(0 to n)(7 downto 0)` for your case. This way you can send and receive the necessary symbols and the last transaction will have the correct _empty_ value.

If you change the code in the receive procedure to exit on `v_sym_cnt = v_symbol_array'length - 1 - v_empty_symbols` I believe there is no way of telling which symbols are valid/invalid in the received data_array (for the last 32-bits), whereas if you use the symbol size elements, your received data_array will only contain valid symbols.

As for the request to check that _empty_ is 0 if it's not the last word in a packet, the Avalon-ST specification mentions the following: _The sink only checks the value of the empty during active cycles that have endofpacket asserted._ So there is no need to check it except during the last cycle.

whiteway11890, 2023-11-13:
Sorry for the horribly long delay - I got pulled to some different projects at work, then I had some personal things come up. 

I should actually be getting back to this in the next week or two. I'll let you know. Thanks again!

UVVM, 2024-07-18:
Hi,
Has this issue been resolved?

whiteway11890, 2024-07-18:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Unfortunately the project I was working on was scrapped and the projects I am now working on don't require those functions. Hopefully I can get back to it someday, but likely not soon.

Sent via the Samsung Galaxy S10, an AT&T 5G Evolution capable smartphone

closed 2024-07-29 by UVVM