UVVM/UVVM #204: Does SBI BFM only support launching data/cmds on falling edge? Can an option be added to t_sbi_bfm_config type to launch/status on either clk edge?

Clemons11, 2023-04-06:
Im using the sbi bfm and I noticed my sims are only launching the commands/data on the falling edge of the clock i pass it and see no options in the `t_sbi_bfm_config` type . The `sbi_write` procedure asserts the command signals after `wait_on_bfm_sync_start` as shown below:

```
wait_on_bfm_sync_start(clk, config.bfm_sync, config.setup_time, config.clock_period, v_time_of_falling_edge, v_time_of_rising_edge);

cs    <= '1';
wena  <= '1';
rena  <= '0';
addr  <= v_normalised_addr;
wdata <= v_normalised_data;
```

and within the `wait_on_bfm_sync_start` procedure it exits right after `wait until falling_edge(clk)`, therefor always launching commands on the falling edge. Can a config option be created to launch on either clock edge?

The current design I am working on has a simple bus interface that is pretty similar but since commands/data is launched on the BFM falling edge and my user logic operates on the rising edge I had to configure the readback check to have a fixed delay as a workaround. Otherwise the `sbi_read` happens on the falling edge as well and my user logic still has half a cycle before the read data is valid.
 ```
use_fixed_wait_cycles_read => true,
fixed_wait_cycles_read     => 1,
```
 But I would prefer it to be a standard single cycle. from rising edge to rising edge. Also, im pretty new to collaborating on public repos, is this something I could help modify, any info is helpful, thank you.

UVVM, 2023-04-12:
Hi, we have some problem understanding this scenario.
As we understand it, your design is positive edge triggered, and if so - then setting the input signals to the SBI interface on the falling edge will just allow a good setup time. The same applies for the read acces where data is read on the rising edge. In both cases the procedures will exit on the falling edge - thus allowing a good hold time. Using the clock edges when entering and exiting the BFMs should work fine for positive edge triggered designs, but this scheme will not allow back-to-back accesses. If you want that, then the bfm_sync must be set to SYNC_WITH_SETUP_AND_HOLD, and the setup_time and hold_time must be set in the BFM configuration.
OK?

Clemons11, 2023-04-12:
Thank you, I will revisit this. I think it was an issue with the back to back accesses like you mention. 

Thanks for the additional info on how to do that!

closed 2024-05-02 by UVVM