UVVM/UVVM #206: [Bug] vvc_generator generates uncompilable code

ThiemoClausen, 2023-05-03:
Hi,

when using the vvc_generator to generte a basic VVC without channels and transaction info, the created code includes some errors.

In order to solve the compile issues you need to:

- Add ``AWAIT_COMPLETION`` and ``AWAIT_ANY_COMPLETION`` to ``t_operation`` in ``vvc_cmd_pkg.vhd``
- Change the type of ``command_queue`` and ``result_queue`` to ``work.td_cmd_queue_pkg.t_generic_queue`` in the main ``_vvc.vhd``
- Change the call for initialization of the interpreter to ``work.td_vvc_entity_support_pkg.initialize_interpreter(terminate_current_cmd, global_awaiting_completion);`` in in the main ``_vvc.vhd``

kind regards
Thiemo

UVVM, 2023-05-08:
Thank you.
This is now fixed and will be available in the next release coming soon.

- UVVM Team

UVVM, 2023-09-16:
Fixed in releas v2023.09.16

closed 2023-09-16 by UVVM