UVVM/UVVM #212: Documentation Error in UVVM VVC Framework PDF

Remillard, 2023-09-08:
I had to go back and forth here with the text a few times but have basically convinced myself that there is a pretty significant typo in the UVVM VVC Framework documentation.  

Page 14 : Figure 6
In this table it is giving examples of valid VVC target addresses.  Item 2 reads:
"2.  UART_VVCT, 4, TX                  Instance number 1 of SBI_VVC UART_VVC"

Based on the context of the preceding discussion, I believe that the "4" is not consistent with the description of "Instance number 1".  Additionally I don't see any reason why this address would target the SBI_VVC.

Recommended update:
"2.  UART_VVCT, 4, TX                  Instance number 4 of UART_VVC"

If there's some other interpretation of this line, I suspect some additional discussion of why it is valid might be warranted because based on the simple explanations in the prior paragraph, there's no foundation for this.

EspenTallaksen, 2023-09-09:
Hi, 
You are absolutely right, and we should also mention that this is for channel TX:
"2. UART_VVCT, 4, TX             Instance number 4 of UART_VVC, channel TX"

I notice we should also update the figure caption

Thanks for your feedback :-)

UVVM, 2024-05-02:
This has been fixed in v2 2024.03.08

closed 2024-05-02 by UVVM