UVVM/UVVM #214: Documentation Error: bitvis_vip_avalon_st

Remillard, 2023-09-13:
Document: avalon_st_vvc_QuickRev.pdf

In table "VVC entity generic constants" several of the entries like GC_CHANNEL_WIDTH and GC_DATA_ERROR_WIDTH mention the following: "Note: If the [signal in question] signal is not used, refer to description in Section 5."

Traversing to Section 5, this section is titled: "Activity watchdog" and does not seem to have any discussion of setting up the module with ports unused.

I'm not entirely sure what to do with unused signals in an `inout` mode port of a record, however setting them to 'Z' seems pretty safe.  I would imagine there are no driver contention issues with that.

I end up with a port mapping as shown below:
```
    ----------------------------------------------------------------------------
    -- Avalon Streaming VVCs for In and Out
    ----------------------------------------------------------------------------
    avalon_st_vvc_1 : entity bitvis_vip_avalon_st.avalon_st_vvc
        generic map (
            GC_VVC_IS_MASTER => TRUE,
            GC_DATA_WIDTH    => 32,
            GC_INSTANCE_IDX  => 1)
        port map (
            clk                              => clk,
            avalon_st_vvc_if.channel         => (others => 'z'),
            avalon_st_vvc_if.data            => wr_din,
            avalon_st_vvc_if.data_error      => (others => 'z'),
            avalon_st_vvc_if.ready           => not(full),
            avalon_st_vvc_if.valid           => wr_en,
            avalon_st_vvc_if.empty           => (others => 'z'),
            avalon_st_vvc_if.end_of_packet   => 'z',
            avalon_st_vvc_if.start_of_packet => 'z'
        );
```

Hopefully that's adequate.  I believe the notes in the quick references should probably be updated with the appropriate information on what to do if some signals in a bus are unused just in case it is not the above solution.

Remillard, 2023-09-13:
Though actually it seems like declaring the instantiation that way does not compile and seems to be much better to define the record with the widths on those unused channels and then only assign the record entries that are being used.  This suggests having a correct Section N that discusses how to declare and invoke the VVC with some signals being unused would indeed be a good documentation section to have.

UVVM, 2024-08-19:
The Avalon-ST documentation has been converted to HTML format and the description has been improved.
You can check it out here: https://uvvm.github.io/

closed 2024-08-20 by UVVM