UVVM/UVVM #217: Reference error in generic_sb_pkg

stfarley, 2023-10-26:
Using UVVM v2023.03.21 I get a runtime error in GHDL

UVVM_ALL/bitvis_vip_scoreboard/src/generic_sb_pkg.vhd:546:14:error: type of variable "vr_sb_queue" referenced before its elaboration
    variable vr_sb_queue         : sb_queue_pkg.t_generic_queue;

This is identified in GHDL issue # 2510

UVVM, 2023-10-30:
Thanks, we will check this.

eminakgun, 2024-03-03:
I receive the same error with GHDL when I elaborate the testbench. Other than that, it runs fine when I run example compile scripts in vip packages. What might be the problem?

eminakgun, 2024-03-03:
I've resolved the problem by just switching to GHDL-3.0.0. I was running ghdl-mcode-11  version 4.0.0-dev.

UVVM, 2024-03-11:
Hi,

The runtime error in GHDL has been fixed in the latest release 2024.03.08
It would be great if you can update the GHDL issue.

closed 2024-04-17 by UVVM