UVVM/UVVM #220: await_completion wrong timeout behaviour

albydnc, 2023-11-20:
I have set up a testbench using Avalon Stream VVCs. The receiver expects the packets and then I call `await_completion` to be sure everyting has been received correctly. I set the timeout to 1 millisecond since there is significant latency between receiver and sender.

When I try to run the test without the await, I am able to see the receiver receiving packets correctly in the waveform, however when I add the await, the test fails with the following error:
```
UVVM: =========================================================================================================================================================================
UVVM: ***  ERROR #1  ***
UVVM:     38259.375 ns   AVALON_ST_VVC,1
UVVM:                    avalon_st_expect(30 words/240 sym, ch:0) while executing avalon_st_receive=> Failed. Timeout while waiting for valid data. 'Receiving 30 words'  [6]
UVVM: 
UVVM: Simulator has been paused as requested after 1 ERROR
UVVM: *** To find the root cause of this alert, step out the HDL calling stack in your simulator. ***
UVVM: *** For example, step out until you reach the call from the test sequencer. ***
UVVM: =========================================================================================================================================================================
```
Here is the the part of tb where VVC calls are done.
```vhdl
    -- TEST : Encode Decode behaviour
    if run("loopback_encode_decode") then
      log(ID_SEQUENCER, "Add expected fixed size " & to_string(N_PACKETS) & " packets to Recv Queue");
      for i in 0 to N_PACKETS loop
        avalon_st_expect(AVALON_ST_VVCT, AURORA_RX_IDX, MAX_MSG(MAX_MSG'length -1 downto i), "Receiving "&to_string(MAX_MSG'length -i));
      end loop;
      log(ID_SEQUENCER, "Sending " & to_string(N_PACKETS) & " packets");
      for i in 0 to N_PACKETS loop
        avalon_st_transmit(AVALON_ST_VVCT, AURORA_TX_IDX, MAX_MSG(MAX_MSG'length -1 downto i), "Transmitting msg "&to_string(MAX_MSG'length -i));
      end loop;
    end if;
    -- UVVM cleanup
    await_completion(VVC_BROADCAST, 1 ms, "Wait for all the VVCs to finish");
    report_alert_counters(FINAL);
```

UVVM, 2023-12-05:
Hi
Sorry for the delay in getting back to you on this. 
Does the code you posted include all VVC calls made in your tb? Would it be possible for you to share your full log output?

UVVM, 2025-03-20:
Closed due to missing feedback.