UVVM/UVVM #227: File with UVVM libraries inter dependencies

m-kru, 2024-01-23:
Is there any file in the repository with UVVM libraries inter dependencies? It can be really simple plain text file, for example:
```
uvvvm_util
uvvm_vvc_framework
bitvis_vip_avalon_mm: uvvm_util
bitvis_vip_avalon_st: uvvm_util
...
```
The benefit of having such a file was that build systems would be able to easily track dependencies. Currently, build systems do not track verification libraries at all. Usually, the user simply installs simulation libraries in the location specific to the particular compiler. The drawback is that when a dependency graph is generated for any test bench target, the verification libraries are missing in the graph.

The burden of maintaining such a file is minimal. The file would have a very simple format that builds systems could use to bootstrap build system specific format. Library files are not included in the dependency file because it is assumed that libraries are installed and are not recompiled on each simulation run. The purpose of the file would be to solely describe UVVM inter library dependencies.

UVVM, 2026-10-27:
Hi @m-kru,

As of today's release, every UVVM library has a FuseSoC `.core` file (e.g. `bitvis_vip_spi/bitvis_vip_spi.core`), plus an aggregate `uvvm.core` in the root. They are generated from the compile order files and the library clauses in the sources, so they stay in sync.

The `depend:` list in the `rtl` fileset of each `.core` file holds the inter-library dependencies you asked for, e.g.:

    name: uvvm:uvvm:bitvis_vip_spi:6.0.0
    filesets:
      rtl:
        ...
        logical_name: bitvis_vip_spi
        depend:
        - uvvm:uvvm:bitvis_vip_scoreboard:2.3.4
        - uvvm:uvvm:uvvm_util:2.22.0
        - uvvm:uvvm:uvvm_vvc_framework:2.12.16

The files are plain YAML, so build systems that don't use FuseSoC can still read the dependencies. The `tb` filesets also list the extra libraries
needed by our own testbenches, so use only the `rtl` dependencies for the libraries themselves.

See https://uvvm.github.io/fusesoc.html for how to use UVVM with FuseSoC.

I'll close this issue, but feel free to reopen it if something is missing.

closed 2026-10-27 by UVVM

m-kru, 2026-10-28:
Great, I do not use FuseSoc anymore.