UVVM/UVVM #228: AXI VVC Slave/Subordinate interface Support.

drowzie, 2024-02-02:
Currently BITVIS_VIP_AXI vvc only supports master/manager interface where it acts as a master towards a slave/subordinate module.
Request is to develop a Slave/Subordinate interface where the VVC will receive axi transactions and verify.
The same support request can apply to SBI VVC as it also only supports master/manager IF.

The reason behind this is the cases where one designs a module which acts as a AXI master/manager. A slave/subordinate VVC can then verify the output from the design under test.

Some ideas to support:
* Receive write data on the axi write data transactions.
* Send read data on axi read transactions.

UVVM, 2024-03-04:
Thanks, anything you could add from your customer project?

drowzie, 2025-09-29:
> Any updates on this issue @drowzie ?

No updates here. No time to look into it.