UVVM/UVVM #231: avalon_mm_bfm_config setup not being respected

Remillard, 2024-03-01:
Good morning,

I'm having a bit of trouble getting the Avalon MM bus to work.  Writing works fine, however reading keeps returning with Avalon MM read response being in error.  I have attempted to set this to false, but it doesn't appear to be behind respected.

My setup:

```
    -- Architecture declarations
    signal avalon_bus : t_avalon_mm_if(
        address(10 downto 0),
        byte_enable(3 downto 0),
        writedata(31 downto 0),
        readdata(31 downto 0)
    );
    signal avalon_bus_config : t_avalon_mm_bfm_config := C_AVALON_MM_BFM_CONFIG_DEFAULT;
```

```
    -- Body declarations
    ----------------------------------------------------------------------------
    -- UVVM Avalon-MM VVC Module
    ----------------------------------------------------------------------------
    avalon_bus_config.use_readdatavalid <= true;
    avalon_bus_config.use_response_signal <= false;
    avalon_mm_vvc_1 : entity bitvis_vip_avalon_mm.avalon_mm_vvc
        generic map (
            GC_ADDR_WIDTH   => 11,
            GC_DATA_WIDTH   => 32,
            GC_INSTANCE_IDX => 1,
            GC_AVALON_MM_CONFIG => avalon_bus_config
        )
        port map (
            clk                     => avl_clk,
            avalon_mm_vvc_master_if => avalon_bus
        );
    i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;
```

```
    -- Usage
    STIMULUS : process
    begin
        await_uvvm_initialization(VOID);
        ...
        avalon_mm_check(AVALON_MM_VVCT, 1, 11X"1", X"00000011", "Checking control word", TB_WARNING, C_SCOPE);
```

As noted, I am setting `use_response_signal` to false per the BFM documentation and this configuration object is being used in the mapping instantiation.  It's possible I need to setup this in the declaration section, however for most of these settings I just want the default.  I do use readdatavalid and I do not use response, so I thought this might be adequate.

Please advise.

Remillard, 2024-03-01:
I think I sorted this.  Instead of what I was doing, I created the configuration as a constant by copying the default text out of the package and then changed the entries I wanted:
```
    constant C_AVALON_MM_BFM_CONFIG : t_avalon_mm_bfm_config := (
        max_wait_cycles          => 10,
        max_wait_cycles_severity => TB_FAILURE,
        clock_period             => -1 NS,
        clock_period_margin      => 0 NS,
        clock_margin_severity    => TB_ERROR,
        setup_time               => -1 NS,
        hold_time                => -1 NS,
        bfm_sync                 => SYNC_ON_CLOCK_ONLY,
        match_strictness         => MATCH_EXACT,
        num_wait_states_read     => 0,
        num_wait_states_write    => 0,
        use_waitrequest          => TRUE,
        use_readdatavalid        => TRUE,   -- Not default
        use_response_signal      => FALSE,  -- Not default
        use_begintransfer        => FALSE,
        id_for_bfm               => ID_BFM,
        id_for_bfm_wait          => ID_BFM_WAIT,
        id_for_bfm_poll          => ID_BFM_POLL
    );
```

So if anyone else runs into trying to get this to work, this is how you'd do it (unless there's an easier less verbose way).

erick166, 2024-03-04:
Hi,

Yes, you need to use a constant to set the configuration as this is a generic in the VVC, so using a signal won't work.
Alternatively, you could use the shared variable from methods_pkg inside your test sequencer the following way:
`shared_avalon_mm_vvc_config(<vvc_idx>).bfm_config.use_readdatavalid := TRUE;`

Remillard, 2024-03-04:
Ahh, thank you.  I was unaware of the second method.  That seems like it might be a little more concise but better for just changing 1 or 2 options like I am doing.

erick166, 2024-03-05:
Could you try moving it after the `await_uvvm_initialization(VOID)`?

Remillard, 2024-03-05:
Okay, yes that seems to have done the trick.  I have the Avalon MM VVC component initialized with the default configuration (`C_AVALON_MM_BFM_CONFIG_DEFAULT`) and then in the text exec, the first lines I have are:
```
    begin
        -- Initialization
        await_uvvm_initialization(VOID);
        shared_avalon_mm_vvc_config(1).bfm_config.use_readdatavalid := TRUE;
        shared_avalon_mm_vvc_config(1).bfm_config.use_response_signal := FALSE;
        system_clk_en <= TRUE;
        avalon_clk_en <= TRUE;
        gen_pulse(system_sync_reset, 10*C_SYSTEM_CLK_PERIOD_TIME, "Pulsed reset signal");
```

And later when I do a Avalon read, it doesn't hang looking for Response.  

Thank you!  There's a lot of moving pieces in this library.  It's difficult to know the "gotchas".

erick166, 2024-03-05:
Great! Yes you should always wait for the uvvm initialization before updating the configuration or running any commands to make sure that the VVCs are initialized.

closed 2024-03-06 by Remillard