UVVM/UVVM #233: avalon_mm_check pipeline error

Remillard, 2024-03-06:
I've got another problem with the Avalon MM VVC and while I'm pretty sure the problem is in the use of the VVC (or setup or what-have-you) I cannot figure out how to wrangle the test library into working.  This is also something I believe is actually functionally valid, however due to the UVVM test library I can't get it to validate. 

In short, I have Avalon reading from a block RAM.  I can watch the address assert, the data become valid, the read data valid assert and all of this be ready for the next clock cycle.  However the sequence of checks in avalon_mm_check is getting shifted by one.

The basic code is:
```
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(0, 10), X"00000000", "Checking log addr 0", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(1, 10), X"0000011A", "Checking log addr 1", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(2, 10), X"00070010", "Checking log addr 2", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(3, 10), X"00000000", "Checking log addr 3", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(4, 10), X"00000000", "Checking log addr 4", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(5, 10), X"0000021C", "Checking log addr 5", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(6, 10), X"00080011", "Checking log addr 6", TB_WARNING, C_SCOPE);
        avalon_mm_check(AVALON_MM_VVCT, 2, to_unsigned(7, 10), X"00000000", "Checking log addr 7", TB_WARNING, C_SCOPE);
```

The values there are the values that reside in the block ram.  From the documentation (i.e. the avalon_mm_vvc_QuickRef.pdf), it seems like I can use `avalon_mm_check` in lieu of the 4 commands for reading back a value (`avalon_mm_read`, `get_last_received_cmd_idx`, `await_completion`, and `fetch_result`) and indeed I've used the `avalon_mm_check` in another section where I read back a single register and it worked fine.  Here since I'm queuing them up one after the other I think something is going wrong.

The warnings I start to get are:
```
# KERNEL: UVVM: ID_BFM                          4385.0 ns  AVALON_MM_VVC,2                avalon_mm_check_response(A:x"001", x"0000011A")=> OK, received data = x"11A". 'Checking log
# KERNEL: UVVM:                                                                           addr 1'  [6]
# KERNEL: UVVM: ID_CMD_EXECUTOR_WAIT            4385.0 ns  AVALON_MM_VVC,2                  ..Executor: Waiting for command
# KERNEL: UVVM: ID_CMD_EXECUTOR                 4390.9 ns  AVALON_MM_VVC,2                  avalon_mm_check(AVALON_MM_VVC,2, x"002", x"00070010") - Will be executed  [7]
# KERNEL: UVVM: ID_CMD_EXECUTOR                 4390.9 ns  AVALON_MM_VVC,2                  avalon_mm_check(AVALON_MM_VVC,2, x"003", x"00000000") - Will be executed  [8]
# KERNEL: UVVM: ID_BFM                          4390.9 ns  AVALON_MM_VVC,2                readdatavalid was active after 1 clock cycles
# KERNEL: UVVM: 
# KERNEL: UVVM: ***  TB_WARNING #2  ***
# KERNEL: UVVM:      4392.859 ns   AVALON_MM_VVC,2
# KERNEL: UVVM:                    avalon_mm_check_response(A:x"002", x"00070010")=> Failed. Was x"0000011A". Expected x"00070010".
# KERNEL: UVVM:                    'Checking log addr 2'  [7]
# KERNEL: UVVM: 
# KERNEL: UVVM: ID_CMD_EXECUTOR_WAIT            4392.8 ns  AVALON_MM_VVC,2                  ..Executor: Waiting for command
# KERNEL: UVVM: ID_CMD_EXECUTOR                 4398.7 ns  AVALON_MM_VVC,2                  avalon_mm_check(AVALON_MM_VVC,2, x"003", x"00000000") - Will be executed  [8]
# KERNEL: UVVM: ID_CMD_EXECUTOR                 4398.7 ns  AVALON_MM_VVC,2                  avalon_mm_check(AVALON_MM_VVC,2, x"004", x"00000000") - Will be executed  [9]
# KERNEL: UVVM: ID_BFM                          4398.7 ns  AVALON_MM_VVC,2                readdatavalid was active after 1 clock cycles
# KERNEL: UVVM: 
# KERNEL: UVVM: ***  TB_WARNING #3  ***
# KERNEL: UVVM:      4400.672 ns   AVALON_MM_VVC,2
# KERNEL: UVVM:                    avalon_mm_check_response(A:x"003", x"00000000")=> Failed. Was x"00070010". Expected x"00000000".
# KERNEL: UVVM:                    'Checking log addr 3'  [8]
# KERNEL: UVVM: 
```

I snipped out reading address 0 because the returned value is zero.  The second address (0x1) is correct and matches.  I can see this in the waveform as well.  The third address (0x2) is correctly data asserted and readdatavalid asserted in waveform, however in the log it looks like there's a sort of command lag here, where the third check is still using the data from the second check.

Do I need to use `await_completion()` with `avalon_mm_check()` here or is there something else going on?  (I can provide a waveform if needed, but I am relatively sure the bus is fine, it's just this interaction with the library is the issue so I'll avoid that for now.)

erick166, 2024-06-27:
Hi,
This seems to be a bug, can you test by adding this wait statement at the beginning of avalon_mm_read_response()?
[image]

Remillard, 2024-06-27:
Unfortunately due to poor business, my company has laid me off and I'm seeking work.  I no longer have access to the tools or code that generated this issue back in March.  (However, if you know of anyone who is looking for a FPGA engineer and willing to accept remote, I'm all ears.)

UVVM, 2024-07-18:
Sorry to hear that. Regarding this issue, it has been fixed in the latest release v2 2024.07.03

closed 2024-07-18 by UVVM