UVVM/UVVM #237: UVVM tag v2_v2024.03.08 breaks old tests

albydnc, 2024-05-07:
Hello,

I have tried to upgrade my tests to [v2_v2024.03.08](https://github.com/UVVM/UVVM/releases/tag/v2_v2024.03.08) however my tests weirdly crash without any info.  
Tests worked fine with [v2023.09.16](https://github.com/UVVM/UVVM/releases/tag/v2023.09.16).

I use VVCs for AvalonST and AXI stream in the tb, tests are run using `vunit`.

This is the verbose output:
```
Starting tb_lib.avst_to_axis_tb.avst_to_axis
Output file: /wrk/vunit_out/test_output/tb_lib.avst_to_axis_tb.avst_to_axis_686df3585fa779b4c96d68d9faffc1a12fc8d74e/output.txt
/wrk/UVVM/uvvm_util/src/rand_pkg.vhd:5507:24:warning: NUMERIC_STD.TO_SIGNED: vector truncated
        v_max_range := to_signed(C_MIN_RANGE, 33);
                       ^
/wrk/UVVM/uvvm_util/src/rand_pkg.vhd:5524:53:warning: NUMERIC_STD.TO_SIGNED: vector truncated
          v_ret_long      := to_signed(v_ret, 33) - to_signed(C_MIN_RANGE, 33); -- Remove offset
                                                    ^
/wrk/UVVM/uvvm_util/src/license_pkg.vhd:52:5:@0ms:(report note): 

*****************************************************************************************************
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
*****************************************************************************************************

/wrk/UVVM/uvvm_util/src/license_pkg.vhd:65:7:@0ms:(report note): 

=====================================================================================================
=====================================================================================================
This info section may be turned off via C_SHOW_UVVM_UTILITY_LIBRARY_INFO in adaptations_pkg.vhd

Important Simulator setup: 
- Set simulator to break on severity 'FAILURE' 
- Set simulator transcript to a monospace font (e.g. Courier new)

UVVM Utility Library setup:
- It is recommended to go through the two powerpoint presentations provided with the download
- There is a Quick-Reference in the doc-directory
- In order to change layout or behaviour - please check the src*/adaptations_pkg.vhd
  This is intended for personal or company customization

License conditions are given in LICENSE.TXT
=====================================================================================================
=====================================================================================================

UVVM: ID_CONSTRUCTOR                     0.0 ns  CLOCK_GENERATOR_VVC,0          VVC instantiated.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  CLOCK_GENERATOR_VVC,0          Command queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  CLOCK_GENERATOR_VVC,0          Result queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR                     0.0 ns  AVALON_ST_VVC,0                VVC instantiated.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AVALON_ST_VVC,0                Command queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AVALON_ST_VVC,0                Result queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR                     0.0 ns  AXISTREAM_VVC,3                VVC instantiated.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AXISTREAM_VVC,3                Command queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AXISTREAM_VVC,3                Result queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CMD_INTERPRETER_WAIT            0.0 ns  AXISTREAM_VVC,3                  ..Interpreter: Waiting for command
UVVM: ID_CMD_INTERPRETER_WAIT            0.0 ns  AVALON_ST_VVC,0                  ..Interpreter: Waiting for command
UVVM: ID_CMD_INTERPRETER_WAIT            0.0 ns  CLOCK_GENERATOR_VVC,0            ..Interpreter: Waiting for command
UVVM: ID_CMD_EXECUTOR_WAIT               0.0 ns  CLOCK_GENERATOR_VVC,0            ..Executor: Waiting for command
UVVM: ID_CMD_EXECUTOR_WAIT               0.0 ns  AVALON_ST_VVC,0                  ..Executor: Waiting for command
UVVM: ID_CMD_EXECUTOR_WAIT               0.0 ns  AXISTREAM_VVC,3                  ..Executor: Waiting for command
UVVM: ID_UVVM_SEND_CMD                   0.0 ns  TB seq.(uvvm)                  ->disable_log_msg(VVC_BROADCAST, ID_PACKET_DATA):  [1]
UVVM: ID_UVVM_SEND_CMD                   0.0 ns  CLOCK_GENERATOR_VVC,0          ->disable_log_msg(VVC_BROADCAST, ID_PACKET_DATA):  [1]
UVVM: ID_UVVM_SEND_CMD                   0.0 ns  AVALON_ST_VVC,0                ->disable_log_msg(VVC_BROADCAST, ID_PACKET_DATA):  [1]
UVVM: ID_UVVM_SEND_CMD                   0.0 ns  AXISTREAM_VVC,3                ->disable_log_msg(VVC_BROADCAST, ID_PACKET_DATA):  [1]
UVVM: ID_CMD_INTERPRETER                 0.0 ns  AXISTREAM_VVC,3                  disable_log_msg(VVC_BROADCAST, ID_PACK
fail (P=0 S=0 F=1 T=1) tb_lib.avst_to_axis_tb.avst_to_axis (0.5 seconds)

==== Summary ===============================================
fail tb_lib.avst_to_axis_tb.avst_to_axis (0.5 seconds)
============================================================
pass 0 of 1
fail 1 of 1
============================================================
Total time was 0.5 seconds
Elapsed time was 0.5 seconds
============================================================
Some failed!
```

albydnc, 2024-05-07:
> Hi,
> This seems a bit strange. Your test is failing even before any activity
> has even started.
> Which simulator are you using? And would it be possible for you to test
> without using VUnit to check if the test case will finish? An alternative
> you can use is HDLRegression
> <https://github.com/HDLUtils/hdlregression/tree/main> if you want to
> automate the test run.
>
> Br,
> UVVM

I tried with a different simulator (NVC) and first it gave me a more
detailed error, stating there was not enough heap space. Enlarging the heap
solved the issue and the test ran fine.
However, I still can't get it to work in ghdl.

erick166, 2024-05-08:
I think this is related to the Axistream VVC, but is a GHDL issue with the memory allocation, see https://github.com/ghdl/ghdl/issues/2644
You can try reducing C_MAX_TUSER_BITS from 64 to 32 in axistream_bfm_pkg as a temporary fix.

UVVM, 2024-07-18:
The constant C_MAX_TUSER_BITS in axistream_bfm_pkg has been reduced from 64 to 32, and moved to adaptations_pkg to be configurable by the user. This should solve the GHDL memory issue.

closed 2024-07-18 by UVVM