UVVM/UVVM #240: UVVM libraries support in Vivado

aranzaba, 2024-06-01:
Hello,

I am launching simulations from Vivado using QuestaSim and UVVM, but the only way that I found to use the VIP libraries is to duplicate the `uvvm_vvc_framework/src_target_dependent` inside of each VIP. See image below.

[image]

The problem is that Vivado does not allow to include the same file in different libraries. Without duplicating the `uvvm_vvc_framework/src_target_dependent` packages in each VIP, Vivado is not able to find the target dependent packages in "work" library, as these are inside the uvvm_vvc_framwork library and not inside the VIP library. Find below a screenshot showing this error from Vivado.
[image]

For any chance did anyone found a more elegant solution to achieve this? And in case that not other better solution is found, would be acceptable to update this in the official UVVM repository?

Many thanks,

Nordin

UVVM, 2024-09-12:
Dear Nordin,
We have discussed this in the UVVM team after your additional feedback at FDF in CERN.
We think it makes sense to make some kind of support for this, but maybe not to duplicate all these files in the UVVM repo. So we have decided to make a script to allow the user themselves to generate these duplicates.
Is it sufficient to generate the duplicates, or does this approach also use our compile scripts or file lists?

Regards
The UVVM team

UVVM, 2024-09-26:
@aranzaba : Hi Nordin, did you see question in the last line above?

aranzaba, 2024-10-11:
Hello,

I apologize, for some reason I did not get a notification for the responses above.

The proposed solution sounds good to me, a script to duplicate these files should be enough.

Thank you,

Nordin

UVVM, 2026-08-28:
The script was released in v2 2026.02.14 under the script/ folder.

closed 2026-08-28 by UVVM