UVVM/UVVM #241: Incorrect waitrequest handling in Avalon MM BFM

aghoras, 2024-06-11:
Hello All,
I believe I have found a bug with the Avalon MM BFM:
If the Avalon MM Slave uses *waitrequest* and does not have it asserted when the read signal from the master goes high, the avalon_mm_read_response procedure (and avalon_mm_check_response procedure) will sample the readdata bus before the rising edge of the clock where the slave is expected to see the read signal asserted which is obviously too soon.  The following patch will fix the problem. I will also do a pull request with the patch.

```

diff --git a/bitvis_vip_avalon_mm/src/avalon_mm_bfm_pkg.vhd b/bitvis_vip_avalon_mm/src/avalon_mm_bfm_pkg.vhd
index c2e703c..aa50bc5 100644
--- a/bitvis_vip_avalon_mm/src/avalon_mm_bfm_pkg.vhd
+++ b/bitvis_vip_avalon_mm/src/avalon_mm_bfm_pkg.vhd
@@ -555,6 +555,12 @@ package body avalon_mm_bfm_pkg is

     -- Handle read with waitrequests
     if config.use_waitrequest then
+      -- Wait until the next rising edge of the clock to check the wait request.
+      -- This will cause the avalon_mm_read_response and avalon_mm_check_response to sample
+      -- the data on the next rising edge of the clock.
+      -- Otherwise, those functions will read the data on the same clock edge as the read signal
+      -- is sampled by the DUT which is obviously too early.
+      wait until rising_edge(clk);
       for cycle in 1 to config.max_wait_cycles loop
         if is_waitrequest_active(avalon_mm_if, config) then
           wait until rising_edge(clk);

```

erick166, 2024-06-27:
Hi,
It seems correct what you are reporting. However, the fix that you made might cause some issues when using pipelined transactions. It might be better to add the wait in the read response, can you test by adding this wait statement at the beginning of avalon_mm_read_response()?

[image]

aghoras, 2024-06-28:
I can try that.. but I'm afraid of adding an extra clock cycle to the case where waitrequest is asserted prior to read. Do you think that will be a problem?

erick166, 2024-06-30:
True, the extra cycle might not be compliant with the timing of the interface defined in the specification.
Then I think we can use your patch for this bug and add an extra fix for the pipeline problem in #233.

UVVM, 2024-07-18:
This issue has been fixed in the latest release v2 2024.07.03

aghoras, 2024-08-30:
I hate to do this but I owe it to the community:

When I filed the original bug, I had checked with Intel regarding their Avalon MM wait request behavior. I started down this path because Intel's FIFO BFM was not working with the UVVM Avalon Memory map master.

Response from Intel:
1. The agent (or slave) is supposed to assert the waitrequest signal asynchronously when it detects the read signal as asserted if it cannot accept data or needs more time to prepare the data. This assertion can happen immediately upon the read request without waiting for a clock edge.

2. If the agent does not need to assert waitrequest for the current operation, such as when reading from a non-empty FIFO, it is not required to drive read data before the next rising clock edge. The agent should drive readdata synchronously with the clock, such that the data is available on the readdata bus at the next rising edge after the read request.

3. The behavior you've observed with the Altera Avalon MM FIFO BFM, where it clears waitrequest when the FIFO is not empty and asserts it when the FIFO is drained, is consistent with synchronous behavior. The readdata is driven after the first rising edge of the clock following the assertion of the read signal, which is the expected behavior for a synchronous module.

4. If the agent does not need to assert waitrequest, the expected behavior would be: a. The host asserts read and drives the address at point a b. The agent drives the data on readdata at the next rising clock edge (point b) c. The host samples readdata at the subsequent rising clock edge (point c) and lowers read d. The agent ignores the read signal at point c since the transaction is complete

Conclusion After Testing With Hardware
Intel's answer is wrong: The fabric does not wait an extra cycle before sampling the readdata bus. In the figure below, the NIOS V sees a value of 0x1 and not 0x2 

[image]

**The UVVM behavior was correct and the salve should either asynchronously drive waitrequest or drive readdata before the next rising edge after the read signal is asserted.** This figure from the Avalon Memory Mapped Interface Specification explains the transaction a little better.

[image]

The correct transaction without waitrequest is
[image]

Master drives address and read at a
The slave drives the data on read_data before the next rising edge of the clock
Master samples the data at c
A burst read would look like:
[image]

A read bust from signal tap:
[image]

_**As a result, I need back out this fix from the baseline**_

UVVM, 2024-09-12:
We already included this, so we now need to check whether to revert this.
Comments are welcome.