UVVM/UVVM #244: Improve readability of log messages

jud43, 2024-06-14:
Hi

I have developed over 100 complex FPGA's in the last 25 years. for these projects I have unknowingly used a similar concept to UVVM with a product called "Testbench_Toolbox". I I am now trying to start using your library.

In your documentation you say that one benefit is 'readability'; however i feel that some of UVVM log messages could be easier to read.please consider the following two examples 'check_value' and 'await_change'

**CHECK_VALUE**
 in Aldec UVVM version  "v2 2021.05.26" if i use 'check_value' like this:
`check_value (spi_ncs, '1', FAILURE, "SPI_nCS is high");`
i see in the console:
`check_value() => OK, for std_logic '1' (exp: '1'). 'SPI_nCS is 1'`

I believe at some point the formatting has changed; because the PowerPoint doc "Simple_TB_step_by_step" page 16 shows the displayed format as
`"check_value (sl 0) => OK. irq2cpu must be default inactive"`
so if we were to use the _same_ format as the powerpoint  the line would be displayed as:
`"check_value (sl 1) => OK. SPI_nCS is high"`
this is much more readable.

**AWAIT_CHANGE**
my TB sequencer calls 'await_change' like this:
`await_change(spi_clk,      0ns, 1*C_SPI_CLK_PERIOD, FAILURE, "SPI_CLK is continuosuly active");`
I see in the console:
```
await_change(std_logic, 0 ns, 2000 ns) => OK. Condition occurred after 20 ns. 'SPI_CLK changes state'
```
**OBSERVATIONS**
I do not understand why

- the 'type' is included in the log message??
- the positive acknowledgement is buried in the log message
- the log messaging cannot be clearer?

**LOGGING SUGGESTIONS**
for example the **await_change LOG format** could be
`verify that <message>, after <min_time> but before <max_time> =>it changed after <actual time> => <RESULT>`
so it would now display on the console as:

```
verify that 'SPI_CLK changes state', after '0ns' but before '2000ns`. => It changed after 20ns => OK
```

and the **check_value LOG format** could be:
`verify that <message> <exp_value>=><RESULT>`
so it would now display on the console as:
`verify that 'SPI_nCS is '1' =>OK`

UVVM, 2024-07-09:
Hi,
and thanks for your feedback. You have a lot of good points here.

We have discussed your input in the UVVM team and we have concluded that we could make some good improvements based on your feedback. Just now, there is vacation time, but I guess we will conclude on how to approach this soon after the vacation. 

There is of course no single correct way of doing this, so we need to evaluate a bit the pros and cons between a couple of potential solutions. We will however most probably continue to include parts of the procedure call in the log. We do agree that with really good user messages there may in many cases be no need to include parts of the command, but from experience and looking at how most users write their command messages, we see that including parts of the command in the transcript is essential in order to allow proper traceability between the test sequencer and the transcript.

We still believe though that we can make significant transcript improvements based on your very good and well examplified feedback. Thanks a lot :-)

-- The UVVM team

EspenTallaksen, 2024-07-14:
Hi,

I'm going on vacation ow, but we will evaluate this after the vacation.

I do however not think we will prioritise a double set of messages via a switch in the adaptation package.
This is quite a bit of work and will make the code less readable. But we will discuss this in August.

With AMD/Xilinx not even supporting VHDL-2008, we are a bit sceptical to including VHDL-2019 support yet, but this - and other VHDL-2019 features will be evaluated within the near future.

-- The UVVM team

sibeov, 2024-07-15:
> Hi,
>
> I'm going on vacation ow, but we will evaluate this after the vacation.
>
> I do however not think we will prioritise a double set of messages via a switch in the adaptation package. This is quite [...]

My impression and experience is that the support for VHDL-2008 is present (still not full) in latest versions of Vivado. When it comes to the `Lite` versions of Quartus however...