UVVM/UVVM #249: Check code quality in a continuous way

racodond, 2024-09-20:
Hi,

I analyzed the VHDL code from your UVVM project with Linty: https://oss.linty-services.com/dashboard?id=uvvm&codeScope=overall

Do not get scared by the number of issues :-)
It's just to provide you with data to browse. I activated all the rules, some are mutually opposed, others are not properly configured with your requirements (such as naming conventions).

We offer Linty for free for open-source projects. I'd be glad to organize a call if you're interested in checking code quality in a continuous way through GitHub actions.
We also offer our VS Code extension for free: https://hdl-designer.linty-services.com/

I'm looking forward to hearing from you.

Thank you

Have a good day

David Racodon
CTO at Linty Services

Paebbels, 2024-09-21:
@EspenTallaksen, @mariuselv, here is my response to such stupid tooling: https://github.com/VLSI-EDA/PoC/issues/74

----
I like checkers, I link linting and I use it a lot for other languages. But the rules in Linty are just against the nature of VHDL and Linty hasn't understood how hardware design and the language works.

UVVM, 2024-09-25:
Hi, thank you both for your input. Feedback and inputs are always appreciated.
We have coding standards for UVVM, and do see the benefits of using linting tools.
As using such a tool would be an investment from our side (several companies are involved in maintaining UVVM) , we would only do that if we were to use that for commercial projects as well. 
Best regards
Marius and Espen

racodond, 2024-09-25:
Hi Marius, Hi Espen,

Many thanks for your reply.

> As using such a tool would be an investment from our side (several companies are involved in maintaining UVVM) , we would only do that if we were to use that for commercial projects as well.

I'm not sure to understand. Could you please clarify?
Do you mean that if you start using Linty for commercial projects, you will then set up Linty for UVVM? Or do you mean something else?

Thank you

Regards,

David

racodond, 2024-09-25:
OK. Got it.
Would you like me to schedule a call with the stakeholders of your companies to discuss your needs and demo Linty?
Shall I remove https://oss.linty-services.com/dashboard?id=uvvm ?

UVVM, 2024-10-17:
Seems to have little priority just now, so stakeholders here will get in touch with you when/if ready for that.
No need to remove it.

closed 2024-12-05 by UVVM