UVVM/UVVM #250: Avalon ST: missing some configuration parameters

albydnc, 2024-09-27:
Hello UVVM devs,

I have been using for a while the UVVM Avalon ST VVCs. I noticed that the BFM is missing some configuration parameters compared to the AXI stream one.
In particular, I was looking for:
```vhdl2008
    valid_low_at_word_num          => 0,
    valid_low_multiple_random_prob => 0.5,
    valid_low_duration             => 0,
    valid_low_max_random_duration  => 5,
    ready_low_at_word_num          => 0,
    ready_low_multiple_random_prob => 0.5,
    ready_low_duration             => 0,
    ready_low_max_random_duration  => 5,
    ready_default_value            => '0',
```

UVVM, 2024-10-04:
Yes, this functionality could be added as well but there hadn't been any requests for it.
If you need it then we can create an internal task to implement it in the near future.

albydnc, 2024-10-07:
Thank you for the response!
I'd like to have it to make the tests more realistic and find some corner cases that can be hidden there.

UVVM, 2025-03-25:
Added to latest release

closed 2025-03-25 by UVVM