UVVM/UVVM #252: Cannot analyse avalon_mm_vvc.vhd with current git versions of nvc and ghdl

yikkrubz, 2024-10-03:
When the current version of uvvm, tag v2024.09.19, is compiled by current versions of nvc or ghdl the file avalon_mm_vvc.vhd does not compile.

Uvvm v2_2024.08.16 can be compiled without issues with both nvc ot ghdl.

Am I missing something?

---------------------------------------------------------
nvc --std=2008 --work=/Users/Peter/.nvc/lib/bitvis_vip_avalon_mm.08 -L/Users/Peter/.nvc/lib -a --relaxed ../src/avalon_mm_vvc.vhd
** Error: type of actual BOOLEAN does not match formal LAST_CMD_IDX_EXECUTED 
          type INTEGER
     > ../src/avalon_mm_vvc.vhd:242
     |
 242 |     update_vvc_activity_register(global_trigger_vvc_activity_register, vvc_status, ACTIVE, entry_num_in_vvc_activity_register, last_cmd_idx_executed, v_cmd_queues_are_empty, C_SCOPE);
     |                                                                                                                                                       ^^^^^^^^^^^^^^^^^^^^^^
** Error: type of actual BOOLEAN does not match formal LAST_CMD_IDX_EXECUTED 
          type INTEGER
     > ../src/avalon_mm_vvc.vhd:250
     |
 250 |     update_vvc_activity_register(global_trigger_vvc_activity_register, vvc_status, INACTIVE, entry_num_in_vvc_activity_register, last_cmd_idx_executed, v_cmd_queues_are_empty, C_SCOPE);
     |                                                                                                                                                         ^^^^^^^^^^^^^^^^^^^^^^
 
-----------------------------------------------------------------
ghdl -a --work=bitvis_vip_scoreboard --std=08 -frelaxed -fsynopsys -Wno-hide -Wno-shared /Users/Peter/Downloads/UVVM/bitvis_vip_scoreboard/src/generic_sb_support_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_scoreboard/src/generic_sb_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_scoreboard/src/predefined_sb.vhd
ghdl -a --work=bitvis_vip_avalon_mm --std=08 -frelaxed -fsynopsys -Wno-hide -Wno-shared /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/avalon_mm_bfm_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/transaction_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/vvc_cmd_pkg.vhd /Users/Peter/Downloads/UVVM/uvvm_vvc_framework/src_target_dependent/td_target_support_pkg.vhd /Users/Peter/Downloads/UVVM/uvvm_vvc_framework/src_target_dependent/td_vvc_framework_common_methods_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/vvc_sb_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/vvc_methods_pkg.vhd /Users/Peter/Downloads/UVVM/uvvm_vvc_framework/src_target_dependent/td_queue_pkg.vhd /Users/Peter/Downloads/UVVM/uvvm_vvc_framework/src_target_dependent/td_vvc_entity_support_pkg.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/avalon_mm_vvc.vhd /Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/vvc_context.vhd
/Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/avalon_mm_vvc.vhd:242:151:error: can't associate "v_cmd_queues_are_empty" with constant interface "last_cmd_idx_executed"
    update_vvc_activity_register(global_trigger_vvc_activity_register, vvc_status, ACTIVE, entry_num_in_vvc_activity_register, last_cmd_idx_executed, v_cmd_queues_are_empty, C_SCOPE);
                                                                                                                                                      ^
/Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/avalon_mm_vvc.vhd:242:151:error: (type of "v_cmd_queues_are_empty" is boolean)
/Users/Peter/Downloads/UVVM/uvvm_vvc_framework/src_target_dependent/td_vvc_entity_support_pkg.vhd:332:14:error: (type of constant interface "last_cmd_idx_executed" is integer)
/Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/avalon_mm_vvc.vhd:250:153:error: can't associate "v_cmd_queues_are_empty" with constant interface "last_cmd_idx_executed"
    update_vvc_activity_register(global_trigger_vvc_activity_register, vvc_status, INACTIVE, entry_num_in_vvc_activity_register, last_cmd_idx_executed, v_cmd_queues_are_empty, C_SCOPE);
                                                                                                                                                        ^
/Users/Peter/Downloads/UVVM/bitvis_vip_avalon_mm/src/avalon_mm_vvc.vhd:250:153:error: (type of "v_cmd_queues_are_empty" is boolean)
/Users/Peter/Downloads/UVVM/uvvm_vvc_framework/src_target_dependent/td_vvc_entity_support_pkg.vhd:332:14:error: (type of constant interface "last_cmd_idx_executed" is integer)
ghdl:error: compilation error

nvc --version
nvc 1.14.0 (1.14.0.r25.g8b6a0378) (Using LLVM 18.1.8)

ghdl --version
GHDL 5.0.0-dev (4.1.0.r238.g0afa60d11) [Dunoon edition]
 Compiled with GNAT Version: 14.1.0
 llvm 18.1.8 code generator

macos 14.7 (intel)

mdykierek, 2024-10-04:
An argument to update_vvc_activity_register is missing. The update_vvc_activity_register has 8 arguments.
  procedure update_vvc_activity_register(
    signal   global_trigger_vvc_activity_register : inout std_logic;
    variable vvc_status                           : inout t_vvc_status;
    constant activity                             : in t_activity;
    constant entry_num_in_vvc_activity_register   : in integer;
    constant executor_id                          : in natural;
    constant last_cmd_idx_executed                : in integer;
    constant command_queue_is_empty               : in boolean;
    constant scope                                : in string := C_VVC_NAME
  );
Only 7 are given:
update_vvc_activity_register(global_trigger_vvc_activity_register, vvc_status, ACTIVE, entry_num_in_vvc_activity_register, last_cmd_idx_executed, v_cmd_queues_are_empty, C_SCOPE);
 update_vvc_activity_register(global_trigger_vvc_activity_register, vvc_status, INACTIVE, entry_num_in_vvc_activity_register, last_cmd_idx_executed, v_cmd_queues_are_empty, C_SCOPE);
executor_id seems to be missing.

UVVM, 2024-10-04:
It seems there was some old code which was not removed during the release.
Can you try commenting out the process p_activity_register_update in line 234?

yikkrubz, 2024-10-04:
Thank you very much for your reply.

After commenting out the process p_activity_register_update the issue is resolved.

mdykierek, 2024-10-07:
It seems that the code released with [v2024.09.19](https://github.com/UVVM/UVVM/releases/tag/v2024.09.19) does not compile with any compiler.
Do you expect fixed code to be re-cheked and tagged with a new tag?

UVVM, 2024-10-10:
The 2024.09.19 release has been reverted and release 2024.10.08 has been published.

closed 2024-10-10 by UVVM