UVVM/UVVM #265: Issue Memory leak

EMAZAC, 2024-11-14:
Dear Sirs,
I think that I found a RAM leak using QuestaSim on Windows.

In my simulation, I'm using the AXI VVC and I'm performing some AXI read every 46 us.

From the attachments, you can see two memory profiling reports, the first one is at 40 ms while the second is at 40.1 ms.

As you can see some code entries are consuming 10 MB of RAM every 100 us.

_axi_channel_handler_pkg.vhd:561
generic_queue_pkg.vhd:988
generic_queue_pkg.vhd:786
axi_read_data_queue_pkg.vhd:122
generic_queue_pkg.vhd:443_

[memory_40000_us.txt](https://github.com/user-attachments/files/17749178/memory_40000_us.txt)
[memory_40100_us.txt](https://github.com/user-attachments/files/17749179/memory_40100_us.txt)

I was not able to find the exact lines of code that produced the leak.

Best regards
Emanuele

EMAZAC, 2024-12-16:
Dear Sirs,
I tested this issue on a different platform:  Questasim 2023 on a Linux PC. The memory leak seems to be present in this case too. 

Best regards
Emanuele

giovanni-emlogic, 2025-03-03:
Hi Emanuele,

After quite some debugging, I am presenting here my findings.

VHDL 2008 states the following about the deallocate() function:

> Procedure DEALLOCATE takes as its single parameter a variable of the specified access type. If the value
> of that variable is the null value for the specified access type, then the operation has no effect. If the value of
> that variable is an access value that designates an object, **the storage occupied by that object is returned to
> the system and may then be reused for subsequent object creation through the invocation of an allocator**. The
> access parameter P is set to the null value for the specified type

It doesn't mandate the release of memory, only to make it available to the OS. At the same time, "may then be reused" indicates that it can still be kept by the tool.

While doing some experimenting, I confirmed that Questasim only allocates memory when needed, but doesn't release it at the exact time that deallocate() is called (this was checked by using the task manager/memory monitor of Windows).

Additionally, I ran a simulation in which I allocated huge chunks of memory to fill all the RAM on my system (Windows OS), then used deallocate() on said chunks. As expected, questasim was still seen as occupying a lot of RAM, but then I opened another app, while still keeping the simulation running, which I forced to take progressively more memory and noticed that the amount of memory occupied by questa started to decrease. My conclusion is that, even tho Questa might take a lot of memory, as long as the memory is deallocated, the OS will be able to use it. If not, the memory is released as soon as the simulation ends. Because of that, while you are correct in stating that the memory is indeed not being released and is accumulating, I think it cannot be considered a memory leak because as soon as the the simulation is closed, all the reserved memory is finally released.

Finally, I think the reason for this is also due to the fact that VHDL does not define a garbage collector, like SystemVerilog, hence it is left up to the simulator to make this kind of decisions.

Best regards
Giovanni

EMAZAC, 2025-03-03:
Hi Giovanni,
I have only two questions to close the issue.

1. Was your analysis performed using the axi package and the generic queue package? Or are you using custom code to verify the Questasim behavior?

2.  During your experiments, did you analyze the QuestaSim memory profile report?
I'm asking this question because I think that your reasoning holds up only if the memory profile report also shows that the memory is not released (after the deallocate()) until another windows app requests the memory. 

Best regards
Emanuele

giovanni-emlogic, 2025-03-04:
Hi Emanuele,

1. I did use a dedicated, more limited test case where I could control the amount of memory allocated. I initially used the axi stream package to transmit many packages, but because I couldn't find the root cause, I tried to challenge my assumption regarding the deallocate function.
2. I did also check the memory profiler which showed different results depending on the command being used. This is how the task manager and the memory profiler compared (fyi I allocated 1GB of data, 7 times using different pointers, then deallocated them one by one). Also to be more precise on the nomenclature,
memory profiler = "profile report -calltree" and then taking "Net memory allocated:"
CMD Proc = is the value obtained by running the same command above and taking the value under "MemUnder(b)" for command_processing.
Capstats = capstats -du+summary
Finally, the data were obtained all in the same run and the entity was compiled with -O0 to avoid optimization (I did notice pre-allocation when not using -O0)

| Task Manager | Memory Profiler | CMD Proc | CapStats - Sesson info | Capstats - HDL | When sampled |
|----------------:|---------------------:|-----------:| -------------------------: | ----------------: | ----------------- |
| 12.4GB      | 52'968                        | 52'490    |       1.86Gb                     | 955Mb             | Before allocate |
| 18.9GB      | 3.820G                       |  21G       |       1.86Gb                      | 7.45Gb             | After allocate, before deallocate |
| 19GB        | 1.115G                        | 14G         |       1.86Gb                     |  7.45Gb            | After deallocate            |

As you can see the memory profiler is all over the place. The "calltree" one reports something, while capstats reports something else. The memory profiler seems also to be deprecated and is suggested to use capstats, ref:

> profile on -m
** Note: (vsim-12129) The memory profiler is depreciated. This usage is ignored. Consider using -capacity and the capstats command.

Although seems to work when clicking the M on the right of the P, which enables the calltree view. So this was the reason for me to focus on the task manager only. Following, the code I used in my test.

```
entity top is
end top;

architecture a of top is
  type t_void is (VOID);

  constant C_SIZE : integer := 25; -- 25=1GB, 50=2GB, 100=4GB
  constant C_10M : integer := 10000000;
  constant C_NUM : integer := 1;
  constant C_UNIT : string := "GB";
  type t_arrInt is array (1 to C_SIZE*C_10M) of integer;

  type mystruct is record    myarr     : t_arrInt;  end record;
  type t_ptr is access mystruct;
  constant c_mystruct : mystruct := (    myarr     => (others => 0)  );

begin
  p_main : process
    variable p1, p2, p3, p4, p5, p6, p7 : t_ptr := NULL;
  begin
    report "Allocating " & integer'image(C_NUM) & C_UNIT;
    p1 := new mystruct'(myarr => (others => 1));
    p2 := new mystruct'(myarr => (others => 2));
    p3 := new mystruct'(myarr => (others => 3));
    p4 := new mystruct'(myarr => (others => 4));
    p5 := new mystruct'(myarr => (others => 5));
    p6 := new mystruct'(myarr => (others => 6));
    p7 := new mystruct'(myarr => (others => 7));
    deallocate(p1);
    deallocate(p2);
    deallocate(p3);
    deallocate(p4);
    deallocate(p5);
    deallocate(p6);
    deallocate(p7);
    std.env.stop;
  end process;
end;
```

EMAZAC, 2025-03-04:
Dear Giovanni,
I understand your methodology and I agree with you. 
Also, this night, I ran a similar test using the axi pkg to allocate/deallocate memory and using an external windows app to request memory. When the external app requests memory, it gets taken from QuestaSim memory.

Therefore there is no memory leak.

Thanks for your time, I will close the issue.

Best regards
Emanuele

closed 2025-03-04 by EMAZAC

giovanni-emlogic, 2025-03-04:
Glad I was able to help :)

Best regards
Giovanni