UVVM/UVVM #271: await_completion with specific cmd_idx fails when VVC is generated without transaction info

llueder, 2024-12-19:
Consider this sequencer snippet:
```
insert_delay (TEST_VVCT, 1, 1 us);
await_completion(TEST_VVCT, 1, 1 ms); --> works
await_completion(TEST_VVCT, 1, get_last_received_cmd_idx(TEST_VVCT, 1), 1 ms); --> timeout
```
Expectation: no failure.
Observation: the second await runs into a timeout.

My thoughts:
`last_cmd_idx_executed` is updated at the end of the executor loop [here](https://github.com/UVVM/UVVM/blob/76556f19019c68e70360214a3bcf7c7c0f4ddb07/uvvm_vvc_framework/script/vvc_generator/vvc_generator.py#L883) and read at the beginning of the next iteration [here](https://github.com/UVVM/UVVM/blob/76556f19019c68e70360214a3bcf7c7c0f4ddb07/uvvm_vvc_framework/script/vvc_generator/vvc_generator.py#L661) with no wait in between. Thus, the finished command is not reported correctly. `wait for 0 ns;` solves the issue.

The provided VVCs here in the repo call `reset_vvc_transaction_info`, which contains a 0 ns wait.
**=> Is it safe to add a 0 ns wait at the end of the executor loop in my VVC, without unintended negative effects within the VVC?**

### Reproduce/mwe
* Clone this repo and cd into it.
* Generate a VVC called "test", with no extended UVVM features, 1 channel, single executor.
* Place the attached compile script and toplevel vhd in the repo root directory.
* In modelsim, run the compilation script and simulate

llueder, 2024-12-19:
related to #220? But I don't think so as the avalon stream VVC mentioned there does have the wait.

erick166, 2025-01-09:
What you are reporting is true, if the VVC does not contain transaction info it should have the `wait for 0 ns` at the end of the executor loop to update the `last_cmd_idx_executed`, you can see that the wishbone VVC has this wait statement.
The vvc_generator.py script should be updated to handle this.

UVVM, 2025-01-10:
The bug in the vvc_generator script has been fixed and will be included in the next release.

closed 2025-03-25 by UVVM