UVVM/UVVM #277: Regression - the images, JavaScript, and *.rst.txt files are missing in documentation

mdykierek, 2025-04-17:
Checked the 2025.04.14 tag:
1294 links in 1314 URLs checked. 21 warnings found. 99 errors found.
Among them:
No such file or directory: 'UVVM-2025.04.14/doc/build/html/_sources/func_cov_pkg_methods.rst.txt'
No such file or directory: 'UVVM-2025.04.14/doc/build/html/_images/direct_transaction_transfer_example_B.png'
No such file or directory: 'UVVM-2025.04.14/doc/build/html/_static/jquery.js'

mariuselv, 2025-04-17:
Did a new rebuild and should be working now. IEEE updated as well

mdykierek, 2025-04-17:
Still No such file or directory: 'doc/html**/images/**vip_gpio/checking_stability.png"

mariuselv, 2025-04-17:
It seems that someone forgot to checkin that file. We'll add it as soon as possible.
Again, thank you for reporting this, @mdykierek

mdykierek, 2025-04-17:
The file is there. The link points to wrong directory. Should read `_images` while reads `images`.

mariuselv, 2025-04-17:
Oh, great. I will update later today then.

UVVM, 2025-05-12:
Fixed

closed 2025-05-12 by UVVM