UVVM/UVVM #278: Simulation speed enhancement: Memory pool

EMAZAC, 2025-05-08:
Dear all,
I'm experiencing a speed bottleneck during my simulation.
When I run simulation (0.5 s simulation time) that executes a lot of VVC axi command, I think to have a bottleneck due to the continuous memory allocation in the generic_queue_pkg add procedure.

[image]

I am looking for a solution and maybe using an internal memory pool for the queue could be the solution.
What do you think?

Emanuele

UVVM, 2025-05-12:
Hi,
Are you distributing a high number of commands to the same VVC, so that the command queue is filling up before the commands are executed?

EMAZAC, 2025-05-12:
Hi,
what is an "high number" for you?

In my simulation the maximum number of command sent to the same VVC is lower than 32 and than the TB wait the execution of all commands. Is this a big number for you?

Also, I'm experiencing simulation speed that decrease during time. 
I want to give more information:
at the simulation begin, to run 1 ms of simulation 9 real seconds are needed
at the simulation end (0.5 s simulation time), to run 1 ms of simulation 130 real seconds are needed

I think the problem is caused by memory fragmentation due to the continuous "new" functions in the generic_queue_pkg add procedure and for this reason I was thinking to a memory pool.

Emanuele

UVVM, 2025-05-19:
Hi,

That doesn't sound good. 32 pending commands should not be a problem with a reasonable data word/package.
Would be great if you could provide the UVVM code you are running?

There is kind of a memory pool available in UVVM. You could check this out and see if this could solve your problem just now: https://uvvm.github.io/fifo_collection.html

With the UVVM FIFO you can send just the sub-FIFO number to a VVC to make it get data for transmission or reception from the FIFO. You would then need to make an overload to pass the sub-FIFO index to the VVC rather than the data, and then add a mechanism inside the VVC to fetch data from the FIFO.

(Would still be good to have your code to check out what could be the problem though...)

-- The UVVM team

giovanni-emlogic, 2025-06-09:
I was able to reproduce the test and I can confirm that the TB gets progressively slower. I'll see what can be done.

Regards
Giovanni

EMAZAC, 2025-06-10:
Hi Giovanni,
I don't know whether to be happy or not about the news :-).
I had a talk with my Windows expert in the company and he thinks that this problem is related to a memory fragmentation.  
Let me know if I can help you.

Best regards
Emanuele

giovanni-emlogic, 2025-06-23:
Hi,

I tested the issue with GHDL, but the issue doesn't seem to affect it as much as with QuestaSim. For every 300ms of simulation I got 20m22s, 21m26s, 21m55s (the sim crashed at around 130ms, probably a limitation on GHDL side). While the timings are increasing, I would have expected a higher increase.

NVC, while did compile, did not run the simulation.

I will try next with Riviera PRO.

The idea is to try narrow the issue, that is to understand if it could be simulator specific or related to the code.

One question, as I have limited resources to look into this issue, could you help me understand how critical is it for your work? Does it hinder it considerably?

Best regards,
Giovanni

giovanni-emlogic, 2025-06-23:
Riviera PRO was way more consistent. These are the times for every 60ms of simulation: 55s 57s 55s 55s 55s 55s 56s 51s 53s 53s.

In any case I will focus debugging this on Questa and will come back as soon as I have something.

Best regards,
Giovanni

EMAZAC, 2025-06-25:
Hi Giovanni,
for my project is very critical. For the first time I'm using the UVVM framework to perform the V&V of a project that shall be compliant with the DO-254 DAL B. I have a test procedure that last 4 seconds (simulation time), the first run has taken about 10 days to be done. After I discover the simulation slowdown and splitting the 4 seconds test procedure in 8 procedures each of 500 ms I was able to reduce the simulation time to about 3 days. Unfortunately I can not split more and, at the same time, I discover the problem in one test procedure but the overall simulation for the project involves 77 different test procedures. 
All the previous V&V project in my company has been done with custom TB but to be able to persuade my manager to move all the next project to the UVVM I need to show that this will be a quality and **time** improvement for the projects.

About the issue, I'm sure that there is no bug in the UVVM code, this is a issue due to the simulator but, maybe, changing the way the queue functionality is implemented It is possible to avoid the slowdown and improve all simulator performance.

Best regards,
Emanuele

giovanni-emlogic, 2025-07-09:
Hi Emanuele,

I looked further into the issue.

Regarding the hypothesis of slowdown due to fragmentation, the queues that are called the most (the ones instantiated in axi_vvc) don't actually have more than 1 element at a time. So if there's no list to traverse, I don't see how fragmentation could affect the speed (at least specific to queue management, but it could affect it in some other ways of course). Also, implementing a memory pool in this case would not be of much help I think.

I checked again my simulation speed and I found something interesting. In the following picture we have the amount of seconds (y-axis) to simulate each millisecond (x-axis). There are three noticeable peaks, unclear to me what caused them (I did not touch my pc during the whole sim). Excluding the peaks, we can see that initially to simulate 1ms takes around 15 seconds. Over time this value grows until around 20s, but remains quite stable. So then what I'm thinking is that there's something happening that pushes up the average simulation time, rather than being a steady increase.

Regarding memory usage, highest I've seen was around 500ms and was 1.7GB.

[image]

I wonder if you'd experience the same. This is the code I added to the tb you provided. It will generate a file every 1ms of simulation. Then by looking at their creation time, you can produce a similar chart to mine in a spreadsheet.

```
process
    file outputf : text;
    variable t : time := 0 ms;
begin
    while true loop
        wait for 1 ms;
        t := t + 1 ms;
        file_open(outputf, "./" & to_string(t, 1 ms) & ".txt" , WRITE_MODE);
        file_close(outputf);
    end loop;
end process;
```

I will try to simulate for an even longer period with a better questa license, but I won't have time to look into this before a couple of weeks.

Regards
Giovanni

giovanni-emlogic, 2025-07-10:
[image]

I ran it again with a much faster machine and different license. Here there are no peaks. Instead, we can see that while the average simulation time increases, it is due to the occurrence of slower simulation every 2-3ms, rather than a steady increase.

This simulation was run on a linux machine.

EMAZAC, 2025-07-28:
Hi Giovanni, 
that looks great! I think the memory pool solved the bottleneck in the generic queue; perhaps the remaining 5% is in another piece of code. 

Did you have check the profiler?
Can I try your implementation?

Regards
Emanuele

EMAZAC, 2025-08-20:
Hi Giovanni,
when I talk about a speed improvement using the new code I'm talking about the time needed to simulate 1 ms. 
Considering the average time to simulate 1 ms for the first 200 ms during the previous simulations I have a simulation speed improvement. At the same time the slowdown reduction is clearly visible in my previous post.

Regarding the way to measure the simulation performance I want to propose this approach.

We can remove the file creation process and use a TCL script to run the simulation like this (pseudo code):

```
for i=1:1000 loop
    set start [clock milliseconds]     # store the start time
    run 1 ms                                     # run 1 ms of simulation
    set finish [clock milliseconds]     # store the end time
    puts [ expr {(finish - start)} ]       # print out the time to run 1 ms
end
```

for sure the (finish - start) time can be print in a file for in deep analysis in post-processing. 

What do you think about this approach?

Emanuele

EMAZAC, 2025-09-15:
Hi Giovanni,
your code ([generic_queue2_pkg.txt](https://github.com/user-attachments/files/21513917/generic_queue2_pkg.txt)) resolve the original issue.

Do you know if this implementation will be included in the next UVVM update?

Regards
Emanuele

closed 2026-10-28 by UVVM